pellcorp/creality #119: mjpeg_streamer very often black screen in fluid

Sekilsgs2, 2024-08-18:
mjpeg_streamer very often black screen in fluid  - after updating window - appears again

pellcorp, 2024-08-18:
You using carto probe?  If so this is normal because the camera is stopped fluidd gets a broken stream and no longer tries to download new frames.

Then when camera starts again you need to toggle the camera view to get the browser to retry the camera feed, I don't know if there is a way to configure fluidd to retry or not, it could also be a browser limitation not sure, but it's not anything I can fix

Sekilsgs2, 2024-08-18:
> You using carto probe? If so this is normal because the camera is stopped fluidd gets a broken stream and no longer tries to download new frames.
>
> Then when camera starts again you need to toggle [...]

i'm using bltouch.

And fluid shoiwing black screen randomly I don't understand what it is connected with. This didn't happen on the stock firmware

pellcorp, 2024-08-18:
> > You using carto probe? If so this is normal because the camera is stopped fluidd gets a broken stream and no longer tries to download new frames.
> > Then when camera starts again you need to toggle the camera view to get the browser to retry the camera feed, I don't know if there is a way to configure fluidd to retry or not, it could also be a browser limitation not sure, but it's not anything I can fix
> 
> i'm using bltouch.
> 
> And fluid shows black screen randomly I don't understand what it is connected with. This didn't happen on the stock firmware How revert to old mjpeg streamer - or this didnt help?

Ive never seen with bltouch so not sure, not possible to revert sorry, creality version of mjpg streamer is not easily stopped and started which is needed for simple AF, i will see if there is config options to that might need to be tweaked

pellcorp, 2024-08-18:
I mean you can easily revert back to stock firmware for sure just not stay with simple AF with old mjpg streamer, creality a mess of it so it's not possible to stop and start it directly only via udev which is not convenient for probes like carto which need to be able to stop the camera

closed 2024-08-27 by Sekilsgs2