pellcorp/creality #125: M600 and Filament Management

TrainAss, 2024-08-25:
Hoping to have M600 and some Filament management added. Being able to load/unload filament without doing a print, and filament swaps during prints are handy.

Filament change macros are:

[gcode_macro LOAD_MATERIAL]
gcode:
  LOAD_MATERIAL_CLOSE_FAN2
  M109 S220
  G91
  G1 E80 F180
  LOAD_MATERIAL_RESTORE_FAN2
  #BEEP #enable once beeper support is working

[gcode_macro UNLOAD_MATERIAL]
gcode:
  SAVE_GCODE_STATE NAME=myMoveState
  RESTORE_E_CURRENT
  M109 S220
  M83
  G1 E10 F300
  G1 E-15 F3000
  G1 E-22.4700 F2400
  G1 E-6.4200 F1200
  G1 E-3.2100 F720
  G1 E5.0000 F356
  G1 E-5.0000 F384
  G1 E5.0000 F412
  G1 E-5.0000 F440
  G1 E5.0000 F467
  G1 E-5.0000 F495
  G1 E5.0000 F523
  G1 E-5.0000 F3000
  G1 E-15 F3000
  SET_E_MIN_CURRENT
  RESTORE_GCODE_STATE NAME=myMoveState
  #BEEP #enable once beeper support is working

[M600-support.txt](https://github.com/user-attachments/files/16739071/M600-support.txt)

pellcorp, 2024-08-25:
So we already have _GUPPY_LOAD_MATERIAL and _GUPPY_QUIT_MATERIAL, thinking maybe we could move or replace these and move them to useful_macros.cfg and then change guppyscreen.json to use these new macros as well.

pellcorp, 2024-08-27:
I was looking at what others do with pure klipper, first came to this reddit thread:
https://www.reddit.com/r/klippers/comments/qxjcq2/need_a_m600_macro/

And then to the klipper sample macros:
https://github.com/Klipper3d/klipper/blob/master/config/sample-macros.cfg#L101

It might be possible to repurpose the guppyscreen load filament macros - move them to useful_macros.cfg and change guppyscreen.json to use the new macros.

pellcorp, 2024-08-27:
So there is actually an example of the M600 macro here:
https://github.com/fluidd-core/fluidd-config/blob/master/client.cfg

```
[gcode_macro M600]
description: Filament change
gcode: PAUSE X=10 Y=10 Z_MIN=50
```

So I will play with that

pellcorp, 2024-08-27:
BEEP has been added to project recently.

TrainAss, 2024-08-28:
[M600-support.txt](https://github.com/user-attachments/files/16785670/M600-support.txt)
Here is the M600 I'm currently using. I've updated it and fixed some variables so that it'll work properly. I've been testing it and it's working for pausing, unloading, loading, purging and resuming.

Only thing that people will need to do is put their part fan back up to the appropriate speed once resuming.

TrainAss, 2024-08-29:
[additional_macros.txt](https://github.com/user-attachments/files/16805309/additional_macros.txt)
[M600.txt](https://github.com/user-attachments/files/16805318/M600.txt)

Here is my current M600 and additional macros. You'll need both files for your testing.

pellcorp, 2024-08-30:
Ah cool will take a look on weekend hopefully thanks so much

pellcorp, 2024-10-30:
So something like this might work and takes advantage of the existing fluidd client macros, its untested:

```
[gcode_macro _UNLOAD_FILAMENT]
gcode:
  M109 S{printer['gcode_macro RESUME'].last_extruder_temp.temp|int}
  M117 Unloading filament...
  M83
  G1 E10 F300
  G1 E-15 F3000
  G1 E-22.4700 F2400
  G1 E-6.4200 F1200
  G1 E-3.2100 F720
  G1 E5.0000 F356
  G1 E-5.0000 F384
  G1 E5.0000 F412
  G1 E-5.0000 F440
  G1 E5.0000 F467
  G1 E-5.0000 F495
  G1 E5.0000 F523
  G1 E-5.0000 F3000
  G1 E-15 F3000
  BEEP
  BEEP

[gcode_macro _LOAD_FILAMENT]
gcode:
  M109 S{printer['gcode_macro RESUME'].last_extruder_temp.temp|int}
  M117 Loading filament...
  G91
  G1 E100 F180
  G90
  M400
  BEEP

[gcode_macro _PURGE_MORE]
gcode:
  M109 S{printer['gcode_macro RESUME'].last_extruder_temp.temp|int}
  RESPOND TYPE=command MSG="Purging filament..."
  G91
  G1 E10 F180
  G90
  M400

[gcode_macro M600]
description: Filament Change
gcode:
  {% set y_park = printer.toolhead.axis_minimum.y|float + 5.0 %}
  {% set x_park = printer.toolhead.axis_maximum.x|float - 10.0 %}
    RESPOND TYPE=command MSG="Print paused for filament change!"
  PAUSE X={x_park} Y={y_park} Z_MIN=50
  _UNLOAD_FILAMENT
  M106 P0 S0
  M106 P2 S0
  RESPOND TYPE=command MSG="action:prompt_begin Filament change detected!"
  RESPOND TYPE=command MSG="action:prompt_text A necessary filament change has been detected. Please replace filament, LOAD it and click RESUME button."
  RESPOND TYPE=command MSG="action:prompt_button UNLOAD FILAMENT|_UNLOAD_FILAMENT|secondary"
  RESPOND TYPE=command MSG="action:prompt_button LOAD FILAMENT|_LOAD_FILAMENT|secondary"
  RESPOND TYPE=command MSG="action:prompt_button PURGE MORE FILAMENT|_PURGE_MORE|secondary"
  RESPOND TYPE=command MSG="action:prompt_footer_button CANCEL PRINT|CANCEL_PRINT|error"
  RESPOND TYPE=command MSG="action:prompt_footer_button IGNORE|RESPOND TYPE=command MSG=action:prompt_end|warning"
  RESPOND TYPE=command MSG="action:prompt_footer_button RESUME|RESUME|primary"
  RESPOND TYPE=command MSG="action:prompt_show"
```

closed 2024-11-17 by pellcorp