pellcorp/creality #129: Force bed move when homing to prevent damage

TrainAss, 2024-08-25:
When homing the toolhead, the bed does not move to ensure it is clear of the nozzle. This can cause damage to the bed or nozzle in the event that the printer crashes or some other reason or the nozzle to be too close to the bed.

If possible, whatever commands uses to home, should include a move of 5mm at least to ensure the nozzle is clear of the bed.

pellcorp, 2024-08-25:
So sensorless.cfg does this already, however I am unsure it works in all cases, because if Z has not been homed, I don't know if `G0 Z5 F600` actually does anything.

pellcorp, 2024-08-27:
> You could add in a force move of a couple mm at the beginning.
> 
> `FORCE_MOVE STEPPER=stepper_z DISTANCE=5 VELOCITY=5 ACCEL=5`
> 
> This will slowly move the bed down 5mm. I've put it at the beginning of my "homing override" macro ever since I scratched up a bed when it was trying to home.

I even noticed it in the sensorless.cfg for stock creality firmware, my main concern is what happens if for whatever reason the bed is at its lowest point of travel, any force move is going to potentially damage th bed.

So they have this in the sensorless.cfg:

```
{% else %}
    {% if printer['gcode_macro xyz_ready'].z_moved|int == 0 %}
      {% if printer.print_stats.z_pos|float <= 20.0 or printer.print_stats.power_loss == 1 %}
        FORCE_MOVE STEPPER=stepper_z DISTANCE={printer["gcode_macro PRINTER_PARAM"].z_safe_g28} VELOCITY=10
      {% else %}
        FORCE_MOVE STEPPER=stepper_z DISTANCE=0.1 VELOCITY=10
      {% endif %}
      SET_GCODE_VARIABLE MACRO=xyz_ready VARIABLE=z_moved VALUE=1
    {% endif %}
```

I don't understand what `printer.print_stats.z_pos|float <= 20.0`

closed 2024-09-01 by pellcorp