pellcorp/creality #135: Suggestion for new End Print macro

TrainAss, 2024-08-28:
The following end print macro is my suggestion. It will not only move the toolhead to a safe position based on what was just printed, but it will also turn off all heaters and motors once it's moved.

```
[gcode_macro END_PRINT]
gcode:
    SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=0
    BED_MESH_CLEAR                                                                  # clear current mesh        
    SET_SKEW CLEAR=1
    
    M400 ; wait for buffer to clear
    G92 E0 ; zero the extruder
    G91 ; relative positioning
    
    # Set safe speeds
    {% set zVelocity = printer.configfile.settings.printer.max_z_velocity|default(15)|int %}
    {% set maxVelocity = printer.configfile.settings.printer.max_velocity|default(200)|int %}
    {% set zVelocityAdjusted = (0.95 * zVelocity * 60)|int %}
    {% set maxVelocityAdjusted = (0.95 * maxVelocity * 60)|int %}
    
    # Get Boundaries
    {% set max_x = printer.configfile.config["stepper_x"]["position_max"]|float %}
    {% set max_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
    {% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}

# Check end position to determine safe direction to move
    {% if printer.toolhead.position.x < (max_x - 20) %}
    {% set x_safe = 20.0 %}
    {% else %}
    {% set x_safe = -20.0 %}
    {% endif %}
    {% if printer.toolhead.position.y < (max_y - 20) %}
    {% set y_safe = 20.0 %}
    {% else %}
    {% set y_safe = -20.0 %}
    {% endif %}
    {% set lift_height = 25.0 %}
    {% if printer.toolhead.position.z < (max_z - lift_height) %}
    {% set z_safe = lift_height %}
    {% else %}
    {% set z_safe = max_z - printer.toolhead.position.z %}
    {% endif %}
    
    G0 X{x_safe} Y{y_safe} F{maxVelocityAdjusted} ; move nozzle to remove stringing
    G0 Z{z_safe} F{zVelocityAdjusted} ; move nozzle up
    TURN_OFF_HEATERS
    M107 ; turn off fan
    G90 ; absolute positioning
    G0 X15 Y280 F3600 ; park nozzle at rear
    M84 ; Turn off motors
    M117 Print complete
```

pellcorp, 2024-08-31:
Hmm, just noticed this end print is for a K1 Max, I think this is bad ` G0 X15 Y280 F360`

TrainAss, 2024-08-31:
Can you do a macro for regular k1? Since your installer already applies the right files depending on printer.

closed 2024-11-02 by pellcorp