pellcorp/creality #140: Will not home after Print

Face-WASPS, 2024-09-02:
Issue with Sensorless homing after a completed print.

The issue was with the the _Tool_Park_Pause_Cancel. This is uses a Macro for Client_Variable, which needs to be copied into the Printer.cfg. Did this and although it worked it gave me a different error.

Need anything else let me know on discord. Face_WASPS
[20240902-073521.zip](https://github.com/user-attachments/files/16833811/20240902-073521.zip)

pellcorp, 2024-10-20:
this makes no sense at all, the Client_Variable is defined in start_end.cfg

pellcorp, 2024-10-20:
maybe try again with a factory reset and see if the issue persists, as stuff has been moved around a bit since sept 2

pellcorp, 2024-11-17:
So this seems to be peculiar to booty gantry users.   I will investigate changing the installer to configure a rear parking location 10mm back from limits to hopefully avoid this.

closed 2024-11-17 by pellcorp