pellcorp/creality #141: klipper_tmc_autotune

000AXR, 2024-09-03:
addition of support klipper_tmc_autotune
https://github.com/andrewmcgr/klipper_tmc_autotune

pellcorp, 2024-09-10:
So I am working on a change to the install.sh so it works on a K1, once that is working on my fork, you can try it and let me know if you can get it working based on the readme from that project.

https://github.com/pellcorp/klipper_tmc_autotune

I won't have time to do much more than this at the moment.

pellcorp, 2024-09-10:
```
wget --no-check-certificate -O - https://raw.githubusercontent.com/pellcorp/klipper_tmc_autotune/main/install.sh | bash
```

pellcorp, 2024-09-22:
https://github.com/andrewmcgr/klipper_tmc_autotune/pull/200

LifeDay, 2024-09-24:
I gave this a shot and got the following error message:
```
-sh: bash: not found
Connecting to raw.githubusercontent.com (185.199.108.133:443)
writing to stdout
```

I'm running a rooted system (https://github.com/Guilouz/Creality-Helper-Script-Wiki). Not sure if that changes things.

LifeDay, 2024-09-24:
Running with sh seemed to work:
`wget --no-check-certificate -O - https://raw.githubusercontent.com/pellcorp/klipper_tmc_autotune/main/install.sh | sh`
```
Connecting to raw.githubusercontent.com (185.199.109.133:443)
writing to stdout
-                    100% |*****************************************************************************************************************************************************|  2425  0:00:00 ETA
written to stdout

======================================
- Autotune TMC install script -
======================================

[DOWNLOAD] Downloading Autotune TMC repository...
Cloning into '/usr/data/klipper_tmc_autotune'...
remote: Enumerating objects: 781, done.
remote: Counting objects: 100% (491/491), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 781 (delta 468), reused 427 (delta 426), pack-reused 290 (from 1)
Receiving objects: 100% (781/781), 217.87 KiB | 1.06 MiB/s, done.
Resolving deltas: 100% (484/484), done.
[DOWNLOAD] Download complete!

[INSTALL] Linking extension to Klipper...
[POST-INSTALL] Restarting Klipper...
no need to update custom config files
old_version: 1030
new_version: 1030
```

LifeDay, 2024-09-24:
Had a similar issue with the path. Had to hardcode the path as:
`path: /usr/data/klipper_tmc_autotune`

LifeDay, 2024-09-27:
I think my K1 Max has different motors than what's currently in the database. I've added the following lines to my  motor_database.cfg file:
```
[motor_constants cse14hra2l410a-01]
# K1 MAX EXTRUDER MOTOR
resistance: 3.3
inductance: 0.004
holding_torque: 0.19
max_current: 1.00
steps_per_revolution: 200
```
and the following lines to printer.cfg:
```
[autotune_tmc extruder]
motor: cse14hra2l410a-01
```

When I save and restart, I receive the following error message:
`Internal error during connect: module 'extras.tmc' has no attribute 'TMCtstepHelper'`

@pellcorp  Have I done something incorrect on installation?

pellcorp, 2024-10-06:
> I think my K1 Max has different motors than what's currently in the database. I've added the following lines to my motor_database.cfg file:
>
> ```
> [motor_constants cse14hra2l410a-01]
> # K1 MAX EXTRUDER [...]

This is only for Simple AF, its not supposed to work for a normal K1 unless you install bash via entware

LifeDay, 2024-10-07:
Thanks for the clarification!

Tombraider2006, 2024-10-29:
`root@K1-12E3 /root [#] opkg install bash
Installing bash (5.2.21-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/bash_5.2.21-1_mipsel-3.4.ipk
Installing libncursesw (6.4-3) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libncursesw_6.4-3_mipsel-3.4.ipk
Installing libncurses (6.4-3) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libncurses_6.4-3_mipsel-3.4.ipk
Installing libreadline (8.2-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libreadline_8.2-1_mipsel-3.4.ipk
Configuring libncursesw.
Configuring libncurses.
Configuring libreadline.
Configuring bash.
root@K1-12E3 /root [#] wget --no-check-certificate -O - https://raw.githubusercontent.com/pellcorp/k
lipper_tmc_autotune/main/install.sh | bash
Connecting to raw.githubusercontent.com (185.199.108.133:443)
writing to stdout
-                    100% |****************************************************|  2425  0:00:00 ETA
written to stdout

======================================
- Autotune TMC install script -
======================================

[DOWNLOAD] Downloading Autotune TMC repository...
Cloning into '/usr/data/klipper_tmc_autotune'...
remote: Enumerating objects: 791, done.
remote: Counting objects: 100% (501/501), done.
remote: Compressing objects: 100% (70/70), done.
remote: Total 791 (delta 473), reused 436 (delta 431), pack-reused 290 (from 1)
Receiving objects: 100% (791/791), 223.78 KiB | 732.00 KiB/s, done.
Resolving deltas: 100% (489/489), done.
[DOWNLOAD] Download complete!

[INSTALL] Linking extension to Klipper...
[POST-INSTALL] Restarting Klipper...
no need to update custom config files
old_version: 1026
new_version: 1026
root@K1-12E3 /root [#]`

pellcorp, 2024-11-06:
> ```
> root@K1-12E3 /root [#] opkg install bash
> Installing bash (5.2.21-1) to root...
> Downloading http://bin.entware.net/mipselsf-k3.4/bash_5.2.21-1_mipsel-3.4.ipk
> Installing libncursesw (6.4-3) to root...
> Downloading http://bin.entware.net/mipselsf-k3.4/libncursesw_6.4-3_mipsel-3.4.ipk
> Installing libncurses (6.4-3) to root...
> Downloading http://bin.entware.net/mipselsf-k3.4/libncurses_6.4-3_mipsel-3.4.ipk
> Installing libreadline (8.2-1) to root...
> Downloading http://bin.entware.net/mipselsf-k3.4/libreadline_8.2-1_mipsel-3.4.ipk
> Configuring libncursesw.
> Configuring libncurses.
> Configuring libreadline.
> Configuring bash.
> root@K1-12E3 /root [#] wget --no-check-certificate -O - https://raw.githubusercontent.com/pellcorp/k
> lipper_tmc_autotune/main/install.sh | bash
> Connecting to raw.githubusercontent.com (185.199.108.133:443)
> writing to stdout
> -                    100% |****************************************************|  2425  0:00:00 ETA
> written to stdout
> 
> ======================================
> - Autotune TMC install script -
> ======================================
> 
> [DOWNLOAD] Downloading Autotune TMC repository...
> Cloning into '/usr/data/klipper_tmc_autotune'...
> remote: Enumerating objects: 791, done.
> remote: Counting objects: 100% (501/501), done.
> remote: Compressing objects: 100% (70/70), done.
> remote: Total 791 (delta 473), reused 436 (delta 431), pack-reused 290 (from 1)
> Receiving objects: 100% (791/791), 223.78 KiB | 732.00 KiB/s, done.
> Resolving deltas: 100% (489/489), done.
> [DOWNLOAD] Download complete!
> 
> [INSTALL] Linking extension to Klipper...
> [POST-INSTALL] Restarting Klipper...
> no need to update custom config files
> old_version: 1026
> new_version: 1026
> root@K1-12E3 /root [#]
> ```
> 
> I installed bash and installed the script but I still get an error `Internal error during connect: module 'extras.tmc' has no attribute 'TMCtstepHelper'`

So this was never tested with helper script or the version of klipper from Creality, I only ever tested this with the version of klipper we run with Simple AF.

pellcorp, 2024-11-06:
So at this point, tune tmc steppers won't be bundled with Simple AF, I have opened a PR on the original project to support a K1, but its unlikely to be merged as it only supports a K1 running Simple AF, seems like there might be issues with it when running plain old klipper and helper script, but I don't have the time or motivation to investigate helper script issues.

closed 2024-11-06 by pellcorp