pellcorp/creality #158: Add filament sensor control into start and end print macro

TrainAss, 2024-09-18:
Add the following in to the start print macro to ensure the filament sensor is enabled before a print starts.

`SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=1`

--

Add the following to the end print macro to shut it off between prints

`SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=0`

pellcorp, 2024-09-22:
I am happy to make the change, but I don't understand why its necessary, I thought the filament sensor was enabled all the time, what is bad about that?

pellcorp, 2024-09-22:
I will add this but make it configurable in case someone needs to turn this off

pellcorp, 2024-09-22:
it still won't disable it for CANCEL_PRINT though

pellcorp, 2024-09-22:
OK i have merged this, but it won't disable filament sensor in cancel print, not sure how best to do that.

closed 2024-09-22 by pellcorp

TrainAss, 2024-09-22:
In the cancel print macro add 

SET_FILAMENT_SENSOR SENSOR=filament_sensor ENABLE=0

Which will disable it.

As for why this is necessary, having it on all the time can cause a false runout trigger if you're doing any sort of testing or calibrations or any situation where you're bypassing the sensor.