pellcorp/creality #160: Mainline klipper?

MrmysteryV, 2024-09-22:
Does this mod automatically change creality's old version of klipper to mainline? 
Thanks

pellcorp, 2024-09-23:
It is kind of mainline. it is my fork of klipper which is rebased onto mainline every few months, last rebase was late august.

But there are some features specific to K1 that are not in mainline, so its mainline+ some creality specific stuff, like support for their GD32 chips which I do not think will ever be merged to mainline.

Plus my klipper fork has the creality specific firmware bits and pieces as well.

closed 2024-09-23 by pellcorp

MrmysteryV, 2024-09-29:
> It is kind of mainline. it is my fork of klipper which is rebased onto mainline every few months, last rebase was late august.
>
> But there are some features specific to K1 that are not in mainline, [...]

Thanks for the reply. Does that mean I can install klipperscreen and things like happy hare, as well as being able to do a larger than 6x6 mesh?