pellcorp/creality #164: BIQU Microprobe Mounting

MrmysteryV, 2024-10-01:
Hello, I am intending to mount a BIQU microprobe to my k1 max and I worked out that if the probe is turned 180 degrees, you can get a slighly lower offset on x (about 4 mm). I haven't managed to test this model yet, but I have remixed yours and was wondering if my version could be better and what changes I would have to make to the firmware. Thanks

https://www.printables.com/model/1024792-k1-biqu-microprobe-v2-mount-lower-offset

omahena, 2024-10-03:
Hi!

I have experimented with this a while ago. I can't remember with confidence
what the results were. The probe being closer to the printhead increases
the bed surface that can be probed and that is certainly a good thing. But
if I remember correctly there was a downside. This way the probe was more
exposed to heat from the print head.

Please note that the precision of the probe drops with higher temperature
and more importantly, it changes as you heat it up. I initially kept it as
far away from the bed as possible by lowering the bed to the lowest
location during the 15-minute preheating phase. But then the several-minute
long probing of the entire bed would cause the probe to heat up and
progressively lose precision causing problems in the mesh. If I remember
correctly the issue was more pronounced when the probe was closer to the
head. Hmm but the printhead couldn't have been on during probing (no need).
There is also a max operating temperature defined by the manufacturer. I
may have opted to keep it further away to avoid damage to the probe due to
heat exposure. You could try using a thin insulator of some sort.

I am sorry, I can't remember the details, so I encourage you to experiment.
After I wasted 6 months of my life on the K1 Max, I gave up on it a while
back. I now only do PLA prints at the same bed temp and simply reuse a mesh
I made months ago. I would need to replace all the components of the
printer to turn it into a printer that I thought I was buying from the
information advertised. This means I need a different printer. :/

Good luck. Keep us posted on your findings.

pellcorp, 2024-10-20:
wow - this looks really interesting, for sure I will have a look at it when I get a chance and if all is well, can change it to be the default and have the default config work with your mount.

pellcorp, 2024-10-29:
> Hello, I am intending to mount a BIQU microprobe to my k1 max and I worked out that if the probe is turned 180 degrees, you can get a slighly lower offset on x (about 4 mm). I haven't managed to [...]

I mean you just have to change the offset in microprobe.cfg that is it

pellcorp, 2024-11-06:
I won't have a chance to test this at any point in the near future so I am closing this issue.  

If you want users of Simple AF to use your mount be sure to include what changes they should make to microprobe.cfg and microprobe-k1.cfg / microprobe-k1m.cfg to make it easier for them.

closed 2024-11-06 by pellcorp