pellcorp/creality #190: Load filament button ignores selected temperature

dprossner, 2024-10-31:
Regardless of what temperature I set for loading and unloading filament, the printer heats up to 240 degrees which is the maximum option. This is the same as issue  #68 on the guppyscreen repo but updating to the lastest build of guppyscreen does not fix the issue as it does on stock firmware

pellcorp, 2024-11-02:
Executed from guppy screen right?

pellcorp, 2024-11-02:
Ok so for whatever reason guppyscreen behaves differently if the load and unload macros are named something other than the default, I mean I consider this a bug myself, but I can restore the old macros to work around this issue.

So for example this code:

```
if (load_filament_macro == "_GUPPY_LOAD_MATERIAL") {
        const char *temp = lv_btnmatrix_get_btn_text(temp_selector.get_selector(),
                                                     temp_selector.get_selected_idx());
        const char *len = lv_btnmatrix_get_btn_text(length_selector.get_selector(),
                                                    length_selector.get_selected_idx());
        ws.gcode_script(fmt::format("{} EXTRUDER_TEMP={} EXTRUDE_LEN={}", load_filament_macro, temp, len));
      } else {
        ws.gcode_script(load_filament_macro);
      }
```

So in Simple AF we actually  re-use the existing macros LOAD_MATERIAL and UNLOAD_MATERIAL

I need to test this PR but I think it should solve the issue:
https://github.com/pellcorp/creality/pull/197

dprossner, 2024-11-02:
yep that fixed it. thanks!

pellcorp, 2024-11-02:
Ok, I have just merged this so you can update with:

```
/usr/data/pellcorp/k1/installer.sh --update-repo
/usr/data/pellcorp/k1/installer.sh --update
```

closed 2024-11-02 by pellcorp