pellcorp/creality #222: getting error: Must home axis first when calibrating btteddy current

dprossner, 2024-11-10:
when manually adjusting the toolhead to be 2mm off the build plate, it seems to hit a soft limit and go back  up automatically even when the printer has just been power cycled, and when running BTTEDDY_CURRENT_CALIBRATE, the paper test will throw a Must home axis first error when trying to babystep

dprossner, 2024-11-10:
the bigger issue seems to be the soft limit. it keeps jerking the toolhead up when trying to lower it close to the bed without showing anything in the console

pellcorp, 2024-11-10:
I have reverted the force move change, I am thinking there is an issue with it, apologies if so.

dprossner, 2024-11-10:
the issues regarding the must home error seem to have been fixed but it still jerks the toolhead up when getting close to the bed despite using force move in fluidd. Makes it extremely difficult to move 2mm above the bed and absolutely impossible to run the paper test

dprossner, 2024-11-10:
i think part of the problem may be that when x and y are homed, the printer automatically assumes the z position is 3mm instead of it continuing to show the axis being unhomed

dprossner, 2024-11-10:
even using SET_KINEMATIC_POSITION z=100 it still refuses to move close to the bed during the paper test. is the sensor providing feedback to prevent crashing when it should be staying out of the way to find a z offset?

dprossner, 2024-11-11:
z lead screws being unsynced was the cause of the jerking up. fixed by ramming bed to bottom of printer. paper test works fine now

closed 2024-11-11 by dprossner