pellcorp/creality #224: mcu error with btt eddy firmware

dprossner, 2024-11-10:
getting error:

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-304-g50aacb57
MCU(s) which should be updated:
mcu: Current version v0.12.0-295-g6bee7f5f7
eddy: Current version v0.12.0-347-ga91d8a66f
nozzle_mcu: Current version v0.12.0-295-g6bee7f5f7
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'eddy': Unknown command: i2c_modify_bits

when using the btt firmware in the repo on the latest update of simpleaf. 

tried factory resetting from multiple scripts, reflashing the btteddy probe with the given uf2 via multiple methods, even reflashed the mainboard using the method made for unbricking and still no dice.

Weirdly, switching to the eddy firmware from ballaswag's guppymod fixed the issue but i'm still running into the problems on issue #222

dprossner, 2024-11-11:
To add context I have run the script updater which says all mcus are up to date and fully power cycled the printer multiple times

pellcorp, 2024-11-11:
So btt eddy cannot be flashed on the k1 with Simple AF, I know the k1_mod has the capability, Simple AF does not unfortunately, you need to flash it according to the guide in the wiki

https://github.com/pellcorp/creality/wiki/Simple-AF-BTT-Eddy#btt-eddy-firmware

closed 2024-11-11 by pellcorp