pellcorp/creality #243: Merge recent Klipper commit that adds support for Y-Axis twist compensation

awp0, 2024-11-14:
I noticed that Klipper very recently added support for Y-axis twist compensation. For me this may be the missing link preventing me from a perfect first layer, and I'd really love to see it merged into Simple-AF. 

X-axis compensation works really well, but I'm still left with first-layer ripples (nozzle too close) in the front of my plate, which I can only get rid of by adjusting Z-offset to the point where the nozzle is too far in the back of the plate. I think Y-axis compensation would solve this. 

Following is the commit that adds Y-Axis compensation to Klipper:
https://github.com/Klipper3d/klipper/commit/4f3a7fd227ce89dfffdd2b7fc205db410b9a24d5

And a matching commit that ties it together with X-Axis compensation to combine X and Y axis calibration at the same time:
https://github.com/Klipper3d/klipper/commit/38bf6f2693816be08df43fac722dbcfea21238bc

Thank you!!

pellcorp, 2024-12-10:
So I have a new klipper branch which has been rebased on latest klipper, I need guinea pigs

```
/usr/data/pellcorp/k1/installer.sh --klipper-repo klipper jp_simplified_master
```

Switching back is easy:

```
/usr/data/pellcorp/k1/installer.sh --klipper-repo klipper master
```

pellcorp, 2024-12-15:
So we are actually going to rebase on latest klipper in early 2025, then you will have access to this feature.

closed 2024-12-15 by pellcorp