pellcorp/creality #324: Calibrate_Cartographer Macro/Calibrate_Scan Macro

DubiousBiscuit, 2024-12-04:
These two Macros should be renamed to something more relatable. Also, probably need to change the instructions to match the final iteration of the macro name...I think Calibrate Touch...As this stands Probe Calibrate isn't referenced anywhere else..This may also need to apply to the scan macro - I have not tried it as I use the touch option for my cart. 
[image]

DubiousBiscuit, 2024-12-04:
I'm not sure but I think the last step from the original wiki isn't working. Calibrate=1
I'm going to walk through it again.

ZeroDotCMD, 2024-12-04:
I haven't used the Wiki version of calibration since I did the macro. So I wouldn't know. this would be a question for pellcorp.

DubiousBiscuit, 2024-12-04:
I'm sorry it's 3:23 AM here, I misspoke. I meant that step is missing from your macro or at the very least doesn't appear to be working. CARTOGRAPHER_TOUCH CALIBRATE=1. Looking at your macro it's there but it didn't execute when I was documenting the steps of your macro.

ZeroDotCMD, 2024-12-04:
I've just ran through and tested the calibration macro. `STEP 5 - CALIBRATE TOUCH` is working on my end.

DubiousBiscuit, 2024-12-05:
I'll re-run it on mine. I had to a clean install so maybe it will work for me now. Thank you.

DubiousBiscuit, 2024-12-09:
it did work on mine after the clean install. I'm closing this now.

closed 2024-12-09 by DubiousBiscuit