pellcorp/creality #349: [EN] Add firmware compatibility for Ender 3 V3/V3 Plus

boromyr, 2024-12-11:
Hello, would it be possible to extend the compatibility for the firmware also to the sister series Ender 3V3? I specify that I am referring only to the firmware, not to the specific model configuration files or bug fixing support, I will take care of that. 
The motherboard is identical, guppyscreen and helper scripts work in both series, the extruder board should be if not the same, very similar, the mcu is a GD32, just verified. I can perform any test myself taking full responsibility. 
Thank you very much.

pellcorp, 2024-12-11:
Unfortunately I don't have the band width to do this, I am stretched enough as it is supporting the K series.

closed 2024-12-11 by pellcorp

boromyr, 2024-12-12:
Could you give me some guidance on how to do it myself? I had already tried to install your mod, adjusted the configuration, and started the guppy screen, but I was getting an error from the extruder, it didn't recognize the different version of Klipper, I don't think Klipper 0.12 was flashed, unfortunately I didn't take a screenshot.