pellcorp/creality #440: [feature] Allows users to disable Guppy Screen

zevaryx, 2025-01-02:
Per the Discord discussion, I already use external screens to manage my K1 Max

Schulli73, 2025-01-18:
I would only like a BBT EDDY installation without all the other things like Guppy etc. because I would like to have only Mainsail as I have it now without ip:4409

Because I like to use the Creality Helper script from there you get almost everything you need

pellcorp, 2025-02-24:
> I would only like a BBT EDDY installation without all the other things like Guppy etc. because I would like to have only Mainsail as I have it now without ip:4409
> 
> Because I like to use the Creality Helper script from there you get almost everything you need

this is the wrong issue to be quoting on, and also never going to happen.   Creality Helper and Simple AF are two completely different things.

pellcorp, 2025-05-11:
I'm closing this, if someone wants to contribute something they can, but its super low priority

closed 2025-05-11 by pellcorp