pellcorp/creality #495: Support Load Cells as a probe option

pellcorp, 2025-01-13:
Based on some excellent work from a simple af fork, a lot of the work has been done for us, but we would still need to
get a version of klipper with the right bed mcu firmware and any missing load cells commits 

https://github.com/Sekilsgs2/creality_pellcorp

pellcorp, 2025-05-07:
> I am very interested in this, and have been looking at it briefly. Now I already have a btt eddy, but maybe it can be combined? Using a load cell for the calibration of another probe would be easier than the paper test.
> 
> I think the option to have to change the entire wiring is a bit excessive. I think it should be perfectly possible to rewrite the driver a bit to read all 4 sensors at the same time.
> 
> Small question: what config is currently used for the MCU firmware? Is it this one: https://github.com/pellcorp/k1-klipper-firmware

I would entertain anything that means we are using a version of klipper that can't be rebased to stock klipper.  Right now we have a few local commits on top of stock klipper but otherwise it's vanilla, and vanilla klipper will support a single load cell not 4.   

Also there is no support for multiple probes in vanilla klipper and so we are not going to try and support a combination of eddy and load cell this would leave to a version of klipper that diverges too severely from upstream

Yes the firmware blobs for the mcus are from the pellcorp/k1-klipper-firmware repo

errolt, 2025-11-22:
What would be required to use the load cells as a z endstop, not a probe? 

I would be nice to use the physical touch for absolute z sense, and eddy for probing bed distortion. So, one probe(eddy) and one z end stop, not two probes.

pellcorp, 2025-11-22:
So the load cells code is not all merged into upstream klipper and even when it is, it will still only support a single load cell which requires people to do a hardware mod.

Honestly I am not overly motivated to support this and regardless the code is not even merged as yet.

pellcorp, 2026-01-05:
I'm closing this issue, the author of the load cell code to klipper has stated categorically its fairly pointless trying to use the k1 stock MCUs with the klipper load cells code so I think this adventure is over.

[image]

pellcorp, 2026-09-30:
> I am very interested in this, and have been looking at it briefly. Now I already have a btt eddy, but maybe it can be combined? Using a load cell for the calibration of another probe would be easier [...]

Better late than never I guess but yes we are currently testing exactly this, the eddy, bltouch or cartographer for scanning, load cells for z offset and with a driver from open centauri cosmos project to see all 4 load cells as a single cell 

https://github.com/pellcorp/creality/issues/1488

pellcorp, 2026-09-30:
> What would be required to use the load cells as a z endstop, not a probe?
> 
> I would be nice to use the physical touch for absolute z sense, and eddy for probing bed distortion. So, one probe(eddy) and one z end stop, not two probes.

Yes we are testing this now 
https://github.com/pellcorp/creality/issues/1488