pellcorp/creality #503: Capture fluidd / mainsail logs

pellcorp, 2025-01-17:
Investigate if there is a way to configure moonraker / fluidd / mainsail to save the console logs somewhere so that for support its possible to grab those logs.

ducati1198, 2025-01-17:
Just A suggestion that I use on a other printer that I have is a script that I put in a USB key and insert it in the printer and fetches all the logs to copy them on the USB key. I’m not sure if it could work with the K1

pellcorp, 2025-01-17:
> Just A suggestion that I use on a other printer that I have is a script that I put in a USB key and insert it in the printer and fetches all the logs to copy them on the USB key. I’m not sure if it could work with the K1

Not really related to this feature request at all, if you are interested in something like this, post it as an new feature request.

pellcorp, 2025-01-17:
So this is not implemented in moonraker, so at this point in time, the feature is dead in the water, I am closing it as WONTFIX cos I can't do anything.

https://github.com/Arksine/moonraker/issues/335

closed 2025-01-17 by pellcorp