pellcorp/creality #520: Request: If its possible the have the hotend go down to 140-150° when pausing the print

ducati1198, 2025-01-23:
If its possible the have the hotend go down to 140-150° it will stop the nozzle to leak when you pause the print. And when resuming it will wait to bring back temperature before moving. Thanks

Not sure if this help take the decision. On my multi color printer with ace pro when a problem occurs with the filament (broken filament let say) the printer pauses and the temperature drop to 150° after fixing the filament problem and press resume it brings back to to temperature will purge a some filament and start to reprint. I know that the k1’s as no poop shoot but you can purge a bit a filament on any side on de bed plate. Just a idea 💡

clutchkick512, 2025-01-23:
This is my pause-resume.cfg(resaved as txt file for github to upload) from a different printer, might work as a base to configure this request.

[pause-resume.txt](https://github.com/user-attachments/files/18519732/pause-resume.txt)

pellcorp, 2025-01-24:
unless this can be done as part of the fluidd / mainsail client variables, I am unwilling to consider it, I just don't have the to replace all that stuff.

pellcorp, 2025-01-24:
I actually wonder if it will be as simple as these changes to start_end.cfg, so a new macro:

```
[gcode_macro ON_PAUSE]
gcode:
   {% set preheat_nozzle_temp = printer["gcode_macro _START_END_PARAMS"].start_preheat_nozzle_temp %}
   M109 S{preheat_nozzle_temp}
```

Then add this to `_CLIENT_VARIABLE`
```
variable_user_pause_macro: "ON_PAUSE"
```

pellcorp, 2025-04-01:
https://github.com/pellcorp/creality/pull/523

pellcorp, 2025-04-13:
this has been released

closed 2025-04-13 by pellcorp