pellcorp/creality #592: [HELP] MCU Protocol error: MCUs not up-to-date?

ricwein, 2025-03-07:
Hello together. I run my K1 with the simpleAF firmware and a cartographer 3d. After powering the machine today i got the appended error-message. What I tried so for:

- updating all packages in the fluidd software-update section - everythin is up-to-date now, klipper shows version `v0.12.0-418-g03096aca`
- the `CHECK_FIRMWARE` macro just states the error below
- updating the firmware itself using the instructions from the [simpleAF docs](https://pellcorp.github.io/creality-wiki/updating/#simple-af)
- reinstalling the the firmware - after the process to script showed "INFO: Your MCU Firmware is up to date!", but the error persists

Connecting of the cartographer to my computer to update its firmware failed. I guess my cable is broken, since the printer itself detects the probe just fine (if you ignore the MCU error...), but my computer just wont show the probe in `lsusb`. But since the Error speaks from `mcu` and `nozzle_mcu` I guess the probe is not the problem here.?

Sorry, if this is a stupid problem, and I'm just to dumb to solve it myself 😅.
Has anyone who is more clever than me an idea on how to fix this?

Greetings, and thank you in advance!

```
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-418-g03096aca-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-412-g8cd3d734f
nozzle_mcu: Current version v0.12.0-412-g8cd3d734f
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

Error during probe mcu identification, check connection:
mcu 'scanner': Firmware constant 'ADC_MAX' not found
```

pellcorp, 2025-03-08:
Please take a look here, its a misleading issue, it just means the carto is either not connected or the serial is wrong:
https://pellcorp.github.io/creality-wiki/cartographer/#error-during-probe-mcu-identification-check-connection

pellcorp, 2025-03-08:
Also you can get more help (and much quicker) if you join our discord
https://pellcorp.github.io/creality-wiki/cartographer/#where-can-i-get-help

closed 2025-03-08 by pellcorp

ricwein, 2025-03-12:
Thank you very much. It looks like my Cartographer is broken. Is it possible to use your simpleAF firmware with the stock load-cell calibration – without a custom probe? I can't find this information in the docs.

I'll leave this ticket closed, since the issue is resolved.

pellcorp, 2025-03-13:
You cannot use simple af with load cells as creality did not release the source for prtouch