pellcorp/creality #737: Override CANCEL_PRINT

pellcorp, 2025-05-09:
So the cancel print provided by the client macros is useful but it puts some annoying constraints on us, it would actually be nicer to just implement our own version of cancel print, in that way we could get rid of the annoying `variable_park_at_cancel_x` and `variable_park_at_cancel_y` and `variable_cancel_retract` which are all used for END_PRINT as well.

So we could instead have our own variables which would be for end and cancel print:

- variable_park_at_cancel
- variable_park_at_end_cancel_y
- variable_park_at_end_cancel_x
- variable_end_cancel_retract

All the logic we have for END_PRINT could be used fairly easily for CANCEL_PRINT as well.

pellcorp, 2025-05-09:
not sure what to do about the use of cancel print when end print is currently executing a cooldown though, could be an issue.

RodentVienna, 2025-05-11:
i agree on the logic and behavior for cancel to behave the same as end.
however i would love to see some kind of user setting variable, that allows to use "unload" at print end or cancel.
A typical, and normal end print would keep the filament in the nozzle (here default retract 5mm), which is perfectly fine for many. I personally prefer to unload filament from the nozzle. there is another use case for "out of the nozzle", which might be "out of extruder! for e.g. M600, and "out of nozzle, but in extruder". These idas are my conclusion so far while trying to understand M600/UNLOAD and Tip forming/shaping. The current setup might however be enough, as we currently have "variable_cancel_retract ", which you want to change to "variable_end_cancel_retract", which could accomodate my use cases. I think I just want select where to unload, and those could be subvariables. with different purpose to seperately use in other macros. bringin in unload/tip forming as part of cancel_end, and making the retract actually configurable from the UNLOAD functionality might be another approach. I will first have to explore that darn M600 tip forming issue, to manifest those theories.

pellcorp, 2025-05-16:
> i agree on the logic and behavior for cancel to behave the same as end. however i would love to see some kind of user setting variable, that allows to use "unload" at print end or cancel. A typical, [...]

I think for more complicated use cases people should just roll their own macros or we can put in some additional custom hooks, I really am not interested in making these macros overly complicated.

pellcorp, 2025-06-03:
I've decided to close this, I don't have the time for going down this rabbit hole

closed 2025-06-03 by pellcorp