pellcorp/creality #760: Use bed heater to heat the eddy before homing

pellcorp, 2025-05-18:
To work around a limitation of the current eddy code (not eddy-ng, classic eddy), we turn the hotend on to encourage the eddy to heat up, but for some users this does not work, consider adding logic to heat the bed as well.

michelszabo, 2025-05-20:
My friend has an observation regarding the bed, it has to be close to the eddy if it is far away, like when the bed is lowered, it doesn't get hot if it is far away.

pellcorp, 2025-05-21:
> My friend has an observation regarding the bed, it has to be close to the eddy if it is far away, like when the bed is lowered, it doesn't get hot if it is far away.

not a lot that can be done in that case other than add a warning to the wiki which I can do, the eddy probe ideally would end up being replaced with the eddy-ng which I think is heading towards being a nicer solution, although I don't think its quite there yet.

pellcorp, 2025-10-15:
If someone wants to provide a PR I will merge it, but Im reluctant to spend any additional time on the classic eddy code, its already taken a lot of my time.

closed 2025-10-15 by pellcorp