pellcorp/creality #821: Is it compatible with Ender 3 V3/V3 Plus?

boromyr, 2025-06-06:
Would it be compatible with Ender 3 V3/V3 Plus or could support also be added for this V3 series since it is already for V3 KE? Available for testing and debugging.
Thank you

pellcorp, 2025-06-08:
So there are a few issues with the Ender 3 V3 as I understood it the main issues are the ADXL support in the nozzle and bed mcus, we don't support the load cell in the KE and we are actually using the existing MCU firmware because we don't have the ability at the moment.

Please join the Simple AF discord because there is another user working on this and they may be able to help you.

https://discord.com/channels/999445685298536638/1369807002678722722

boromyr, 2025-07-10:
Has compatibility been added?