pellcorp/creality #923: [Bug?] print end hit right end when running consecutive prints.

zhongjis, 2025-09-07:
## Just to want to say thank you first
Thank you for the script! It have really made my printing experience awesome!

## My Setup
I am on **Ender 3 v3 KE** with **bltouch default mount** with **latest main branch code (as of 20250906)**.

## My Issue
This issue only happens when I run consecutive prints (no homing all in between). 

When trigger another print right after the previous, the print head will move a bit left then continues going right then keeps hitting right end. On KE the x-axis end pin is located on left so I assume the correct behavior should be going left until pin triggered.

## mitigation
Run homing all after each print could avoid this issue.

I am assuming if there's a missing override for KE and SE. 

## Some Other Random Facts and things I tried
I initially installed SimpleAF couple month ago with the same mount option, at that time everything works.

The issue persist even after I clean reinstall.

_PS: I asked AI and it says the printer could memorize the end location wrong, but my knowledge on this is limited. And have no idea why printer keeps the end location even when the motors were off after each print. Just hope this info helps._

pellcorp, 2025-09-13:
So I can't reproduce this, if you provide the support.zip using the SUPPORT_ZIP macro maybe someone can identify the issue, otherwise please visit the Simple AF discord and a discord member can take a look.   I am taking a break from discord so it would have to be someone else.

zhongjis, 2025-09-14:
sounds good. I already sold my ke to my friend and I restored it to factory firmware as he wont be able to have time dealing with it. guess we can just close this.
I am getting a k1c and looking forward to try simpleAF on that!
Thank you for your script again:)

pellcorp, 2025-09-21:
I have a nebula pad with Simple AF and I could never reproduce this issue at all, so its very odd, shame we could not get to the bottom of it.